Robo3D/Slicing-Profiles #20: Raft Seperation

Ximidar, 2018-03-07:
When printing with onboard profiles I have noticed that the rafts are not coming off, However when using the RoboLCD profile, the rafts separate without much of an issue.

Please update the profiles with these raft settings:
```
raft_airgap: 0.25
raft_airgap_all: 0.0
raft_interface_thickness: 0.27
raft_interface_linewidth: 0.4
raft_line_spacing: 3.0
raft_margin: 5.0
```

For all the printers I have tested on, these seem like good settings for raft seperation.

Here are the current profile raft settings:
```
raft_airgap: 0.26
raft_base_linewidth: 1.0
raft_base_thickness: 0.3
raft_interface_linewidth: 0.6
raft_interface_thickness: 0.2
raft_line_spacing: 3.0
raft_margin: 5.0
```
With these settings I have not been able to get a raft to seperate yet.

AllenMcAfee, 2018-03-07:
@Ximidar the .profile versions have the raft settings you are suggesting, and I've updated the .ini versions to match.  Can you confirm you were using .ini and not .profile?

Ximidar, 2018-03-07:
I'm still seeing the same settings for all the .profile files. I did not use the .ini files, I just dropped in the .profile files.

Ximidar, 2018-03-08:
It turns out that my raft settings don't work on the web dash slicing either. RoboLCD slicer will change the `first_layer_width_factor` key to 100 when a raft is active. On all the profiles this key is set to 300. I will try setting this key to 100 and seeing if that makes the difference.