Robo3D/Slicing-Profiles #28: Primary Extruder Not Accepting "Final Print Temp" Setting

JasonBarretto, 2018-09-08:
# Bug Report

### What did you do?
Set final print temp to 180°

### What did you expect to happen?
Primary hotend to cool to 180°

### What happened instead?
Primary hotend did not cool down until after tool change. 

### What version of Cura are you using?
Cura for Robo version: 3.4.1

JasonBarretto, 2018-09-10:
@AllenMcAfee 
This may have been a one time bug. I have sliced a couple other files since and have not been able to reproduce this.

AllenMcAfee, 2018-09-10:
@JasonBarretto are results looking better?

JasonBarretto, 2018-09-10:
@AllenMcAfee 
Yeah looking better. Still some stringing but I am printing some retraction tests to fine tune the settings and minimize the stringing as best we can.